brown-SSW/brown-solar-charger #4: plan physical dashboard (at least a basic version)

joshua-8, 2021-01-29:
What data and status information do we want to show?
What types of displays are the best for showing that

joshua-8, 2021-01-30:
Estimate how large some numbers might get and make sure displays will be able to show that. Think about what units will be best

joshua-8, 2021-02-02:
status light
battery capacity - display or line of lights
surplus-deficit - display
C02 saved -display (possibly more digits or sci notation)

joshua-8, 2021-03-14:
[image]

sfsw-brown-solar-charger, 2021-03-14:
make it clear that watt-hours generated is cumulative over the life of the station
Make a nice QR code (maybe put our logo in the center?)
One other display (could use some leds to indicate multiple values with one display)