brown-SSW/brown-solar-charger #5: plan website dashboard

joshua-8, 2021-01-29:
after planning, make issues for website components that need to be made, and an issue for exactly what is expected to be in the json data.

joshua-8, 2021-02-02:
hourly power generation-bar graph
daily power generation-bar graph
daily surplus/deficit-bar graph
real time power generation -dial
real time power use-dial
real time battery charge-dial
real time status - text "active" or "out of power"
if the data is unavailable display a warning that the station is down for technical issue
real time Co2 production (miles of driving, gallons/m^3 of natural gas) -numerical counter
real time approximate number of computers charged (estimate power) -numerical counter

vskbellala, 2021-02-07:
I've outlined JS functions to serve the following purposes:
- Update plots and other visuals on a specified interval without refreshing the page. (loadJSON)
- Modify the page to display something or change styles (really anything you can do with html DOM) whenever the user cannot access the .json (isDown)
  - inverse of this function restores the website to its initial, working state when the .json is accessible (isUp)

All this code to modify the page and render plots runs in a function called "callback," which executes every time loadJSON is called. In terms of what to do next, I would like to fully implement status messages/indicators and css/js gauges to display various statistics that wouldn't go on a graph. I also intend to finalize general styling for our plotly visuals over this next week.

Any suggestions for specific aesthetics to have would be greatly appreciated!

closed 2021-08-02 by joshua-8