brown-SSW/brown-solar-charger #7: better graphs for webpage

joshua-8, 2021-01-29:
This should probably be a javascript function.
You can use the drawGraph1 function that's already in the code as a starting point if you want. It displays a simple bar graph of json data, and can be resized, but isn't easily reusable for other graphs and is missing many things like labels and axes.
I'm likely missing something, and feel free to go far outside this structure if you want to or for any reason, but here's some ideas to start with:
**some parameters you might want the function to expect**
name of the canvas to draw on
name of the div that canvas is inside of (for resizing)
data to display (for drawGraph1 I think you'd want to get json_obj.graph1)
color and style options
scale for y axis (or do auto scaling)
labels for axes, and maybe for numerical value of bar graph
maybe have a line or scatter plot option?

And you can ignore all of this if you'd rather figure out how to use someone else's library to make plots!

vskbellala, 2021-01-31:
Using an existing JS based library might be easier to set up and maintain. I suggest [plotly.js](https://plotly.com/javascript/) as I have some experience with its python sibling.
I'll get a demo up and running over this week.

joshua-8, 2021-02-09:
@vskbellala made a functional demo with plotly so I think this issue can be closed

closed 2021-02-09 by joshua-8