brown-SSW/brown-solar-charger #12: energy to co2 conversion

joshua-8, 2021-01-30:
Find a factor to convert watt hours to amount of co2 saved

lrandall521, 2021-02-18:
1 kWh of electricity produced here in Massachusetts accounts for 0.846 lbs of carbon. (0.0023 gallons of gas per hour)
https://www.bostonsolar.us/solar-blog-resource-center/blog/how-much-co2-do-solar-panels-save/

A laptop that is on for eight hours a day uses between 150 and 300 kWh and emits between 44 and 88 kg of CO2 per year (20g CO2 per hour on average)
https://www.energuide.be/en/questions-answers/how-much-power-does-a-computer-use-and-how-much-co2-does-that-represent/54/

lrandall521, 2021-02-21:
[image]

lrandall521, 2021-02-24:
In terms of raw energy conversion (not hourly use):
1 lbs of carbon dioxide = 1.087 kWh (https://www.eia.gov/tools/faqs/faq.php?id=74&t=11)
1 gallon gas = 33.7 kWh (https://en.wikipedia.org/wiki/Gasoline_gallon_equivalent)

lrandall521, 2021-02-24:
Made a bunch of plots: https://docs.google.com/spreadsheets/d/1UoQ_441te4A9Lstv__UHW8SKH3CZwFIRyuwbllHcdvE/edit#gid=0

closed 2021-08-02 by joshua-8