brown-SSW/brown-solar-charger #21: work on program for backend server

joshua-8, 2021-02-14:
I know that isn't a specific issue, this is just to remind us that we need to figure out what needs to be done.

lrandall521, 2021-02-18:
I created a Firebase project for hosting, though I had to use my personal account as I was given the following error with my Brown account:
The operation has failed (Reason: Google Cloud Platform service has been disabled. Please contact your administrator to restore service.)

It took me awhile but I finally got firebase to work. I had to mess around a lot with the git workflow and secrets and overall if this doesn't work I'm going to be very upset :(

Here's a tutorial that I couldn't really make it through:
https://www.youtube.com/watch?v=9kRgVxULbag&feature=emb_logo

joshua-8, 2021-02-20:
Thanks for starting our research into this. firebase is somewhat working as a database now, so I think we can close this issue

closed 2021-02-20 by joshua-8