brown-SSW/brown-solar-charger #25: send email reports

joshua-8, 2021-02-17:
we can discuss changes to this, but I'm thinking something like monthly usage summaries with a dump of that month's detailed logs along with immediate error reports if the station goes offline.

joshua-8, 2021-02-20:
maybe this can be done by a script connected to a spreadsheet?

vskbellala, 2021-02-20:
I like the idea! Are you planning to use a mailing list type system like [mailchimp](https://mailchimp.com/) so multiple people can subscribe to it?

joshua-8, 2021-02-20:
I hadn't thought of that, good idea!

joshua-8, 2021-02-28:
I have made a couple google groups and set up the sfsw.brown.solar.charger gmail (ask me and I'll send you the password) so that if the station emails itself, the message can be forwarded to people who are subscribed to the group. I made one group for regular (maybe monthly) reports on the station's performance, and another group for people who want to receive alerts if the station detects a technical issue that needs to be fixed.
Both groups can be found by going to google groups logging in with the solar charger email, but since I haven't even started the process of testing the system I recommend that no one subscribe yet unless they're the one working on sending emails, since there will probably be a lot of annoying tests. Much later when everything is running smoothly we can tell everyone that they can subscribe to these groups if they want (and we should add the s4sw email).

joshua-8, 2021-07-21:
I think this issue can be closed. I will make some more specific issues for anything left that relates to email reports.

commit d7a83863 (2021-07-21): close #37 close #35 close #28 close #25
closed 2021-07-21 by joshua-8