brown-SSW/brown-solar-charger #27: make google sheet record all data

joshua-8, 2021-02-20:


joshua-8, 2021-02-28:
can this be done by making a periodically running google sheet script that checks the json for new data and adds it to the sheet? Other than logging the daily arrays (and keeping it longer than 24 hours) could it also log some values from live data or even debug?

lrandall521, 2021-02-28:
https://docs.google.com/spreadsheets/d/1-D1cuN569Oko7t2OFXpSDvPqCG6O4iQV2vTHWb0dfH4/edit?usp=sharing

lrandall521, 2021-02-28:
For syncing with firebase: https://websiddu.com/blog/sync-data-from-firebase-to-google-sheets.html

Requires data is stored in cloud firestore first.

sfsw-brown-solar-charger, 2021-03-08:
https://docs.google.com/spreadsheets/d/125tzejewiulYwVp-gOtJQhKFdyUY960JVkXAU_3bh_U/edit#gid=0

https://github.com/brown-SSW/brown-solar-charger/wiki/google-apps-script-for-importing-json-data-into-spreadsheet

closed 2021-03-08 by sfsw-brown-solar-charger